donny681/ESP32_CAMERA_QR #3: ov2640 not initializing with SXGA mode

chegewara, 2018-10-20:
I am trying to initialize camera with mode SXGA or UXGA and i am getting errors:
```
SCCB_Write [ff]=00 failed
SCCB_Write [e0]=04 failed
SCCB_Write [da]=18 failed
SCCB_Write [d7]=03 failed
SCCB_Write [e1]=77 failed
SCCB_Write [44]=0c failed
SCCB_Write [e0]=00 failed
SCCB_Write [ff]=00 failed
SCCB_Write [44]=63 failed
```

FangJiJun, 2019-01-17:
Hello, I have the same problem now,How do you solve this problem?thank you

chegewara, 2019-01-17:
I did not. I was not able to run OV2640 with resolution higher than SVGA if i remember, or i got only artifacts.

FangJiJun, 2019-01-18:
Are you still trying to do this？can you speak Chinese？

chegewara, 2019-01-18:
Sorry i dont speak chinese.

You can try to ask on this repository:
https://github.com/espressif/esp-who

esp-eye board also is using ov2640.

FangJiJun, 2019-01-19:
The problem that caused may be that RESET in the code corresponds  to PWDN of OV2640.
I've got the image of UXGA. you can refer to this code , only need to do is modify the pins,and RESET pin set HIGH.
https://github.com/m5stack/m5stack-cam-psram

chegewara, 2019-01-20:
Thanks, i will check it.

kisel, 2019-02-17:
Had the same issue with UXGA mode. This fork worked for me(/jpg & /jpg_stream work fine): https://github.com/tmrttmrt/ESP32_CAMERA_QR/commit/1192823e1751434340717309931a63b1009c3621

jcyfkimi, 2019-02-18:
> 
> 
> The problem that caused may be that RESET in the code corresponds to PWDN of OV2640.
> I've got the image of UXGA. you can refer to this code , only need to do is modify the pins,and RESET pin set HIGH.
> https://github.com/m5stack/m5stack-cam-psram

There's more than pins, the root cause of this issue is that the SRAM is not enough for SXGA/UXGA frame buffers, it's only 520KB SRAM and by calculation, SXGA resolution needs more than 1200KB as frame buffers, the PSRAM must be involved in the project if you want to achieve large resolution pictures, and also you need to consider the read/write speed of the PSRAM.

akshayvernekar, 2019-03-24:
@kisel I am getting frame allocation error , what is the SRAM on your esp32 ?

Maoweicao, 2019-07-28:
Maybe set Enable malloc on external SPI memory for FreeRTOS to slove this problem,see https://docs.espressif.com/projects/esp-idf/zh_CN/stable/api-reference/system/himem.html and
https://github.com/espressif/esp-idf/tree/v3.2.2/examples/system/himem
I test to slove it success,but the picture aren't what I want,it look like aren't what I want.Maybe should setting camera config again,OV2640 can work normal.
说起来可以通过启用ESP32使用外置SPI完成malloc功能来解决这个问题，具体看上面那两个链接，我试过之后发现可以正常用，但是照片看起来很怪，是那种彩虹条纹的照片，可能是摄像头时间搞得不对。

closed 2019-12-04 by donny681