donny681/ESP32_CAMERA_QR #5: http_server component conflict on build

edgarmove, 2018-11-06:
Hi, on the latest IDF I saw a lot of errors trying to compile, all related to the http server.

I assume that it is a conflict with the espressif http server component with the custom http server of your code.

Just for someone else experimenting this issue, my unelegant workaround was to exclude the Espressif libraries in the Makefile:

EXCLUDE_COMPONENTS := esp_http_server protocomm

It builds now. 

Thanks for your work!

JiriBilek, 2019-01-02:
Hi, for clean build I needed
`EXCLUDE_COMPONENTS := esp_http_server esp_https_server protocomm`

kreuzhofer, 2019-01-14:
I just tried to add you exclude to my Makefile and also tried to add it to component.mk in the main directory. It has no effect for me, the error still resists. What might I be doing wrong?