donny681/ESP32_CAMERA_QR #6: Connecting the Ai-Thinker CAM board to PC

gazialankus, 2018-11-20:
How do I connect the Ai-Thinker CAM board to PC? The links in the [quick start](https://github.com/donny681/ESP32_CAMERA_QR#quick-start) lead to ESP32 boards with a USB port. 

I guess I need to connect the Ai-Thinker CAM board to the USB port of my PC somehow. I have one of [these](https://www.aliexpress.com/item/USB-switch-ESP8266-WIFI-module-board-mobile-phone-computer-wireless-communication-MCU-WIFI-development/32799975353.html) for ESP8266, can I use this for this purpose somehow? 

Thanks.

drJeckyll, 2018-11-27:
Have no idea about your 8266 "device" (may be it can be used, I just don't know and don't see schematics for it).
Better get one of USB-FTDI modules like [this](https://www.aliexpress.com/item/1pcs-FT232RL-FTDI-USB-3-3V-5-5V-to-TTL-Serial-Adapter-Module-forArduino-Mini-Port/32650148276.html) or something similar. Use 3.3V voltage selector.
To program ESP32-CAM you need to connect:
U0R and U0T from ESP32-CAM to TX(0) and RX(I) from FTDI board (or swap if not working), GND and 3.3V also.
Then connect IO0 and IO2 from ESP32-CAM to GND.
make flash will do the rest - then remove IO0 to GND (may be IO2 to GND also), so module can boot from flash.
That's it.
P.S. Be sure you enable CONFIG_SPIRAM_SUPPORT - it is not set by default (Support for external, SPI-connected RAM under ESP32-specific) or there will be no enough memory to run QR demo.

aguaviva, 2018-12-04:
Does the module have that external SPI-connected RAM?

drJeckyll, 2018-12-04:
@aguaviva Can speak for my module, which don't have enough memory to run code without enabling this option. After that it run OK.

raphaelbs, 2018-12-28:
Hi @gazialankus 
I got my Ai Thinker working using a USB to TTL like [this one](https://hobbycomponents.com/1820-thickbox_default/ftdi-usb-serial-port-adapter.jpg).

**Steps to flash**
- `IO0` to ground
- Short board ground to USB ground
- USB `RX` to `U0T`
- USB `TX` to `U0R`
- Power up the board with external power (can be `5v` or `3v3`)

If you don't pull `IO0` to ground the board will boot in `RUN` mode.
Also you need external power because USB ports doesn't have enough juice.

closed 2019-12-04 by donny681