donny681/ESP32_CAMERA_QR #9: Sometimes I get a black square

raphaelbs, 2018-12-28:
Hello,

Just got the [Ai Thinker ESP32-CAM](https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20181228054032&SearchText=esp32+cam).

### What I know

- If `IO2` is not pulled to ground I get the black square bellow in any resolution.
- With `IO2` to ground, I got it working with `QVGA` and `QQVGA`.
- Higher resolutions (> `QVGA`) gives the black square even with `PSRAM`. 

[image]

### What I tried

By default `PSRAM` was disabled so I enabled by adding to `sdkconfig`:

```
CONFIG_SPIRAM_SUPPORT=y

#
# SPI RAM config
#
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_SPIRAM_USE_MEMMAP=
CONFIG_SPIRAM_USE_CAPS_ALLOC=
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
CONFIG_SPIRAM_SIZE=4194304
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=
```

In the boot screen, I could see that `SPI SRAM` was active:

[image]

With `VGA` resolution and `PSRAM` enabled, I verified that `DMA buffer` was allocated correctly:

[image]

The HTTP server seems fine, even though it responds a square:

[image]

9crk, 2019-01-05:
me too. this project was shut down by Ai-Thinker corp.
it's not a good project.

9crk, 2019-01-05:
you can't find this module sale on the Ai-thinker website or taobao shop now.

raphaelbs, 2019-01-05:
They are still available on Aliexpress though.
I'm still working with with this module and will open a repo just for it sometime soon.

bokse001, 2019-01-05:
I have the same problem with a black screen but also with a distorted picture. @raphaelbs I am very interested in what you produce. Seems there are some more implementations out there, not specific for the ESP32-CAM but a different board that use the same basis implementation.

bokse001, 2019-01-06:
I found out that, when you do a completely new build, it works without distortions... But why? Is there a driver mismatch when you compile with a different resolution and only the app gets rebuild?
Also I put in the fix for the fb size and PSRAM.

raphaelbs, 2019-01-06:
@bokse001 just openned the repo. Please check it out and feel free to contribute or discuss about it.

donny681, 2019-01-26:
@raphaelbs @9crk  I will fix the promblem soon.Recently I'm so busy.Maybe you can use the older sdk.

closed 2019-01-26 by raphaelbs