donny681/ESP32_CAMERA_QR #10: Ov2640 module not working!!

searchingforcode, 2019-01-01:
Hi i am using this module for testing https://www.aliexpress.com/item/OV2640-camera-module-Module-2-million-pixel-electronic-integrated-with-jpeg-compression-new-big-promotion/32915971973.html . It doesnt have XCLK pin on header as it says it already has an integrated 12mhz crystal. I have already succeeded with esp32-cam board (small camera one) with this firmware. But this ov2640 camera module not working. Its showing "Camera probe failed with error 0x20001". I think this code refers to connection issue. I have changed the wiring and pin configuration but still no change. My pin configuration is: 

vsync-25
href-33
reset-23
d1-32
d3-12
d5-13
d7-21

scl-26
sda-27
d0-35
d2-34
d4-5
d6-18
plck-19
pwdn-gnd

Thanks. Waiting for replies.

amenimtibaa, 2019-03-18:
Hello, 
Have you found a solution for this ? I am facing the same error with the same module as you

aizukanne, 2019-03-19:
Hello,

I have the same challenge. Did you find a solution?

amenimtibaa, 2019-03-19:
unfortunately, not yet

AuComrad, 2019-09-02:
Hi,
I'm having the same problem. Has anyone found a solution yet?

goebish, 2020-02-12:
I'm having the same issue (ESP32-CAM with Ov2640), I know the module is working fine since I uploaded and used a demo Arduino sketch without any issues.

With ESP32_CAMERA_QR:

I (537) sccb: pin_sda 26 pin_scl 27
I (567) sccb: SCCB_Probe start
E (1827) camera: Camera probe failed with error 0x20001
E (1827) camera_qr_demo: Camera init failed with error 0x20001

Was anyone ever able to get it working with an Ov2640 camera module ?

amenimtibaa, 2020-02-12:
Hey guys, 
So, I figured out a solution that worked well for my ESP WROVER.
[Here](https://drive.google.com/file/d/1RRilG1w3iDGCrUFNJhC3eZOIYvkKtB2I/view?usp=sharing)  is the configuration I used :)