donny681/ESP32_CAMERA_QR #21: ov7670

amirjak, 2019-04-18:
hi
how work with ov7670 without fifo
[image]

donny681, 2019-05-18:
I have yet finished it

amirjak, 2019-05-18:
thanks for answering
goodluck

jjsch-dev, 2019-11-07:
I ported this project to the idf version 4.1 dev. Replace the camera component for the **esp_idf/components/esp32-camera**  that support the ov7670 and hhtp_server for **esp_idf/components/esp_https_server**

I added a conversion function for pgm of 8 bits to RGB bitmap og 24 bits.  

Now the WIFI_SSID and WIFI_PASSWORD are configured in the example connection menu.

https://github.com/jjsch-dev/ESP32_CAMERA_QR/

donny681, 2019-12-04:
@jjsch-dev Thank you ,but I tried your demo,but lack something

jjsch-dev, 2019-12-04:
How can I help you?

donny681, 2019-12-04:
@jjsch-dev I fixed the demo last night.But you can try to suppport  more cameras,such as ov5640

jjsch-dev, 2019-12-04:
I don't have this sensor, if you send me one I will try to add it to the esp camera sensor driver.

closed 2020-05-13 by amirjak