donny681/ESP32_CAMERA_QR #28: error: 'cif_regs'

lesiyu129, 2019-09-19:
Toolchain path: /Users/lesiyu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /Users/lesiyu/esp/esp-idf/requirements.txt are satisfied.

App "esp32-cam-demo" version: 964ba57-dirty
CC build/camera/ov2640.o
/Users/lesiyu/esp/ESP32_CAMERA_QR/components/camera/ov2640.c:208:22: error: 'cif_regs' defined but not used [-Werror=unused-const-variable=]
 static const uint8_t cif_regs[][2] = {
                      ^~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [ov2640.o] Error 1
make: *** [component-camera-build] Error 2

chopin1998, 2019-09-30:
same error

rezaarifandee, 2019-10-28:
Edit the Makefile and add line below after PROJECT_NAME

```
CFLAGS = -Wno-error=unused-const-variable
```

Your Makefile should now looked like this

```
PROJECT_NAME := esp32-cam-demo
CFLAGS = -Wno-error=unused-const-variable
include $(IDF_PATH)/make/project.mk
```

Hope it helps.

closed 2019-12-18 by donny681