donny681/ESP32_CAMERA_QR #33: Guru Meditation Error

esemerino, 2019-11-16:
When I introduce the "make monitor" after the "make flash", the mysis32 take loop message, and inside of the loop, there is an error that says " Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled."

Its the only error in the loop, and the i have proved the camera example, and it works and take image in the ip...

Anybody have ever seen this? Anybody knows solve the problem?
thanks

donny681, 2019-12-04:
I haven't met this problem.Can you provide more detail?

esemerino, 2019-12-04:
Yes...
The program doesn't compile correctly and get a cyclic message, with information about the module  ESP and operations thats makes correctly. But inside of the cyclic message, there is a error that says "guru Meditation......"
its like an unfinished compilation...as if the program doesen't get the whole compilation and it failed in a point of it...
The config is ok, the camera pins are configured correctly

donny681, 2019-12-05:
Please check the power .The current should be over 1A.Can you show your device picture.

closed 2019-12-18 by donny681