donny681/ESP32_CAMERA_QR #35: ESP32 with OV2640 get 0x20001

tobearlabs, 2019-12-04:
I have the MAKERFACTORY M5Stack ESP32 Kamera mit OV2640 (2.0 MP)    
(https://www.conrad.de/de/p/makerfactory-m5stack-esp32-kamera-mit-ov2640-2-0-mp-1888217.html)
with USB-C.

I can flash, everything is running and in the menuconfig i prepared it for OV2640 support.

But I get: 
I (0) cpu_start: App cpu up.
I (446) heap_init: Initializing. RAM available for dynamic allocation:
I (452) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (458) heap_init: At 3FFB91D0 len 00026E30 (155 KiB): DRAM
I (465) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (471) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (477) heap_init: At 40095858 len 0000A7A8 (41 KiB): IRAM
I (484) cpu_start: Pro cpu start user code
I (502) spi_flash: detected chip: gd
I (502) spi_flash: flash io: dio
I (502) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (552) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=8
I (552) I2S: PLL_D2: Req RATE: 39062, real rate: 39062.000, BITS: 16, CLKM: 16, BCK: 8, MCLK: 10000000.000, SCLK: 1249984.000000, diva: 64, divb: 0
I (562) camera_xclk: PIN_CTRL before:3ff,39062
I (572) camera_xclk: PIN_CTRL after:7fff
E (4582) camera_demo: Camera probe failed with error 0x20001

Do yyou have any idea to get it to work with that USB integrated? Thanks!!!!

tobearlabs, 2019-12-04:
More information:

I (360) esp_image: segment 5: paddr=0x000b3d2c vaddr=0x40080400 size=0x19fc0 (106432) load
I (429) boot: Loaded app from partition at offset 0x10000
I (429) boot: Disabling RNG early entropy source...
I (429) psram: This chip is ESP32-D0WD
I (435) spiram: Found 64MBit SPI RAM device
I (438) spiram: SPI RAM mode: flash 40m sram 40m
I (444) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (451) cpu_start: Pro cpu up.
I (454) cpu_start: Application information:
I (459) cpu_start: Project name:     esp32-cam-demo
I (465) cpu_start: App version:      0d87564-dirty
I (470) cpu_start: Compile time:     Dec  4 2019 22:13:55
I (476) cpu_start: ELF file SHA256:  e698264323feb2e0...
I (482) cpu_start: ESP-IDF:          v4.1-dev-1086-g93a8603c5
I (489) cpu_start: Starting app cpu, entry point is 0x40081448

seblee, 2019-12-05:
Note README.md
## Note:
If log debug come out "Camera probe failed with error 0x20001",please check whehter the camera connect to Board or check the pin Num.

donny681, 2019-12-05:
@seblee 强 厉害 赞 真棒

donny681, 2019-12-05:
@tobearlabs absoultely yes,I make sure the demo can work.Please check your camera or pin num.

closed 2019-12-05 by donny681

tobearlabs, 2019-12-05:
@donny681 Thanks for your reply:
My nums are, given by the supplier:
D2 -> IO32
D3 -> IO35
D4 -> IO34
D5 -> IO5
D6 -> IO39
D7 -> IO18
D8 -> IO36
D9 -> IO19

in the pin configuration in "make menuconfig" I added all the values starting in your config with D0 with my D2...up to your D7 (which used my D9 so 19).

I set all the other values regarding to my table...still the ame error with my OV2460.

Any ideas?

tobearlabs, 2019-12-05:
This is the output with the right NUMS:
E (1477) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=12
E (1487) camera_xclk: ledc_timer_config failed, rc=ffffffff
E (5507) camera_demo: Camera probe failed with error 0x20001