donny681/ESP32_CAMERA_QR #36: Help,how can i implement into esp32

Dlayz1, 2020-01-24:
Sorry,im new,may i ask how can i implement and upload this into esp32 with detail guide and explaination?the file in the main directory,i cant find the .ino file need to open it into the arduino and as the description stated that i need the esp idf library too.how can i do these?

donny681, 2020-02-06:
Please check out the introduction.

jenmwms, 2020-03-27:
@Dlayz1 Keep at it! I'm trying to get this working too. I found this for the ESP-IDF:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/. I see in the `Makefile` uses the environment variable IDF_PATH so we will need to install it. Then, I think we run (in command line, or terminal) make esp32-cam-demo?