donny681/ESP32_CAMERA_QR #40: compilation error

poaongithub, 2020-06-02:
hi, @donny681, I got compilation error after clone your code and install "idf.py",  how to solve it? 
Any help appreciated, thanks in advance.
-------------------------
/home/bo/ESP32_CAMERA_QR/main/app_main.c:54:8: error: unknown type name 'esp_ip4_addr_t'
 static esp_ip4_addr_t s_ip_addr;
        ^~~~~~~~~~~~~~
In file included from /home/bo/projects/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:43,
                 from /home/bo/projects/esp/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from /home/bo/projects/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/bo/projects/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/bo/projects/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/bo/ESP32_CAMERA_QR/main/app_main.c:27:
/home/bo/ESP32_CAMERA_QR/main/app_main.c: In function 'app_main':
/home/bo/projects/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip4_addr.h:176:65: error: request for member 'addr' in something not a structure or union
 #define ip4_addr_get_byte(ipaddr, idx) (((const u8_t*)(&(ipaddr)->addr))[idx])
                                                                 ^

ArkYk, 2020-07-06:
I'm not sure,maybe you can watch this url.[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/tcpip_adapter_migration.html](url)

shun-dgut, 2020-08-27:
main/app_main.c  修改为下面这样应该就可以了
//static esp_ip4_addr_t s_ip_addr;  
static ip4_addr_t s_ip_addr;