donny681/ESP32_CAMERA_QR #42: Missing CMakeLists.txt file

kopiking, 2020-07-22:
Hi, I'm unable to build this project or access menuconfig because there is no CMakeLists.txt file in the folder. I've cloned this project into my esp-idf folder (so the current project directory is esp-idf/ESP32_CAMERA_QR). I'm not sure if I'm supposed to generate a CMakeLists.txt file from the project directory, or if I'm supposed to use the CMakeLists.txt from the esp-idf directory.

Any help would be appreciated, thanks

dblatt87, 2020-08-03:
Hello, I like to check if my OV7725 works on my ESP32-CAM. Unfortunately, tutorials for the work with CMakeLists.txt are quite rare, so I stuck in the same place as kopiking...

I made CMakeLists.txt files in each directory as I noticed in the esp-who camera-web-server example, which I got working (at least with OV2640, not with OV7725...).

This fails at the linker:
[3/4] Linking CXX executable ESP32_CAMERA_QR.elf
FAILED: ESP32_CAMERA_QR.elf
cmd.exe /C "cd . && C:\Users\Dome\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\ESP32_CAMERA_QR.elf.rsp  -o ESP32_CAMERA_QR.elf  && cd ."
c:/users/dome/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp32/libesp32.a(cpu_start.c.obj):(.literal.main_task+0x4): undefined reference to `app_main'
c:/users/dome/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp32/libesp32.a(cpu_start.c.obj): in function `main_task':
C:/Users/Dome/Desktop/esp-idf/components/esp32/cpu_start.c:545: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Content of "esp-idf\ESP32_CAMERA_QR\CMakeLists.txt":
############################################
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS ../../../components)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(ESP32_CAMERA_QR)
############################################

Content of "esp-idf\ESP32_CAMERA_QR\main\CMakeLists.txt":
############################################
set(COMPONENT_SRCS "app_main.c" "led.c" "qr_recoginize.c"
                    INCLUDE_DIRS "")

set(COMPONENT_REQUIRES
    camera
    http_server
    quirc
    )

register_component()
############################################

Thanks in advance for any help

jungu009, 2020-08-12:
I add the cmakelists.txt in [this](https://github.com/jungu009/ESP32_CAMERA_QR), You can refer to it

dblatt87, 2020-08-12:
Thanks, compiles and works now on my ESP32-Cam with OV2640, not with OV7725. 
But [this ](https://github.com/espressif/esp-who/issues/151)might be a HW issue...

danielc007, 2020-08-20:
Managed to solved my compilation problem by changing:
static ip4_addr_t s_ip_addr;
to
static esp_ip4_addr_t s_ip_addr;
in app_main.c

closed 2020-11-19 by donny681